ARM-software/CMSIS-Driver #2: i2c multi slave acknowledge polling

bart112233, 2019-05-14:
For acknowledge polling after write command (ex.  24AA02E48 - for waiting write operation done ) is need to call MasterTransmit with params data=NULL and num=0. But implementation  of I2C_MasterTransmit in I2C_MultiSlave.c don't allow this case.

VladimirUmek, 2019-05-16:
Hi,

for this particular device you could implement a workaround by sending 1 byte using MasterTransmit. That extra byte would serve as a dummy Word Address. As a consequence read operations would need to specify (re-set) Word Address again after each write operation.

I agree this is not optimal, but it should work. Do you know other practical use cases for calling MasterTransmit using data = NULL and num = 0?

bart112233, 2019-05-16:
Hi, 

  24AA02E48 have one byte length addresses - i will check your method, thanks.

  No, i don't know any other practical usage of this case - only for eeprom ack poll.

VladimirUmek, 2019-05-22:
Were you able to solve the problem using suggested workaround?

bart112233, 2019-05-22:
Yes, your method works. Many thanks for you help.

closed 2019-05-22 by bart112233