ARM-software/CMSIS-Driver #4: ESP32 BLE Support

yldzmuhammed, 2020-03-05:
Hello and thanks this great job.

Does this driver support BLE functionality?

If does not, will it be?

If never, why?

If in the future, what is the road map?

VladimirUmek, 2020-03-05:
Hello,
BLE functionality is not supported as ESP32 driver is only [WiFi driver](https://arm-software.github.io/CMSIS_5/Driver/html/group__wifi__interface__gr.html) focused on socket communication.

We intend to explore BLE functionality but at the moment there is no decision about supporting it.

yldzmuhammed, 2020-03-05:
OK. How can I help with this? Is there any one interested?

VladimirUmek, 2021-01-18:
Closing this issue since there is no further interest and no short term plans to extend support to BLE.

closed 2021-01-18 by VladimirUmek