ARM-software/CMSIS-Driver #6: ESP32 baud rate change with AT 2.1

xennex22, 2020-12-23:
The board rate change logic does not work with the ESP32 AT firmware 2.0 or 2.1

The reason is that the AT command to test if communications are good at the current baud rate may return an ERROR from a previous AT command at the wrong baud rate.  Since we are just trying to sync baud rates ERROR is as good as OK.

https://github.com/ARM-software/CMSIS-Driver/blob/52078889fcb1eb83ac69cc0f6dff35f9e4c9de50/WiFi/ESP32/WiFi_ESP32.c#L4334

Change to

`/* response might be OK or ERROR (from last bad baud attempt) */`
`if ((ex == AT_RESP_OK) || (ex == AT_RESP_ERROR)) {`

VladimirUmek, 2021-01-18:
Hi,
thanks for reporting this. Unfortunately the driver was not yet validated with any of AT 2.x firmware because we have seen instabilities in ESP32 transfer functions.
Currently the driver is only validated for use with AT 1.2.0.0 which passed the socket transfer validation tests:
https://github.com/ARM-software/CMSIS-Driver/blob/52078889fcb1eb83ac69cc0f6dff35f9e4c9de50/WiFi/ESP32/CMSIS_DV_Results/About.txt#L1-L18

We will take this issue into account and check what needs to be done in order to get the driver running with ESP32 AT 2.x firmware.

xennex22, 2021-01-18:
Thank you.  If I can find AT 1.2.0.0 I will re-test on that version.

If you asked me 30 years again when I was working on AT commands for modems that I would still be using the AT commands in 2021 I would have quit and become a carpenter.

VladimirUmek, 2021-01-18:
The firmware is listed under releases here https://github.com/espressif/esp-at/releases (last release before V2.0).

VladimirUmek, 2021-01-21:
The driver has been updated - the fix you suggested was added and I also tested the driver with ESP32 AT 2.1.0.0 which now passes our validation suite without any failed tests.

The recommended firmware version is now ESP32 AT 2.1.0.0 since it behaves very stable (definitively more stable that 2.0.0.0). Although you maybe already flashed your ESP32 to AT 1.2.0.0 I would now recommend going back to 2.1.0.0.

closed 2021-01-21 by xennex22