ARM-software/CMSIS-Driver #7: ESP32 does not send more than 2048 bytes correctly

xennex22, 2021-01-09:
There is a problem sending more than 2048 bytes via TCP in `ARM_WIFI_SocketSendTo`

The logic to break the transmit data into chunks of 2048 bytes or less does not decrement to original data size.

There's a loop to send to data to the ESP32
https://github.com/ARM-software/CMSIS-Driver/blob/52078889fcb1eb83ac69cc0f6dff35f9e4c9de50/WiFi/ESP32/WiFi_ESP32.c#L3363-L3366

But nowhere does `len` get decremented.  The inner loop counters do get decremented/incremented.

https://github.com/ARM-software/CMSIS-Driver/blob/52078889fcb1eb83ac69cc0f6dff35f9e4c9de50/WiFi/ESP32/WiFi_ESP32.c#L3427-L3428

So if I transmit 2050 bytes rather then sending 2048 bytes then 2 bytes the function just keeps trying to send 2048 bytes until eventually it fails.

xennex22, 2021-01-09:
The return value from `ARM_WIFI_SocketSendTo` has a problem too, as it checks if the last chunk size (max 2048) is equal to the total tx size (may be over 2048)

https://github.com/ARM-software/CMSIS-Driver/blob/52078889fcb1eb83ac69cc0f6dff35f9e4c9de50/WiFi/ESP32/WiFi_ESP32.c#L3444-L3446

And this is done in the inner of two loops, so as soon as one chunk is tx then the rval is set, which exists the outer loop, which then compares the total tx with the rval and fails.

VladimirUmek, 2021-01-18:
Hi,
implementation is a bit confusing due to its complexity, but what happens in line
https://github.com/ARM-software/CMSIS-Driver/blob/52078889fcb1eb83ac69cc0f6dff35f9e4c9de50/WiFi/ESP32/WiFi_ESP32.c#L3364
is only a check when application called the driver with len argument equal to zero. According to [ARM_WIFI_SocketSendTo documentation](file:///C:/Keil/ARM/PACK/ARM/CMSIS/5.7.0/CMSIS/Documentation/Driver/html/group__wifi__socket__gr.html#ga5ddd3dd5f863b8392e4281dd48a28d28) this is allowed to check (pool) if the socket is ready to send data.

Since ESP32 can only send 2048 bytes with a single AT+CIPSEND command, initial count is limited to 2048 bytes:
https://github.com/ARM-software/CMSIS-Driver/blob/52078889fcb1eb83ac69cc0f6dff35f9e4c9de50/WiFi/ESP32/WiFi_ESP32.c#L3367-L3375
which are then loaded into ESP32 internal buffer: https://github.com/ARM-software/CMSIS-Driver/blob/52078889fcb1eb83ac69cc0f6dff35f9e4c9de50/WiFi/ESP32/WiFi_ESP32.c#L3399-L3401
and the total number of byte transferred is incremented while loading buffer:
https://github.com/ARM-software/CMSIS-Driver/blob/52078889fcb1eb83ac69cc0f6dff35f9e4c9de50/WiFi/ESP32/WiFi_ESP32.c#L3427

When the transfer is done and "SEND OK" is received from ESP32, the number of total bytes sent is checked:
https://github.com/ARM-software/CMSIS-Driver/blob/52078889fcb1eb83ac69cc0f6dff35f9e4c9de50/WiFi/ESP32/WiFi_ESP32.c#L3444-L3446
If this check is not true, the whole process will be repeated. Any additional number of bytes left from 2048 is then sent with next AT+CIPSEND command.

xennex22, 2021-01-18:
If you can understand my pseudocode:

```
len = the total number of bytes to transmit
num = 0
while(len > 0) {
  cnt = the minimum of 2048 and the len
  send 'AT+CIPSEND' + cnt
  wait for 'OK'
  while (cnt != 0) {
    n = minimum of data to send and AT buffer size
    n = number of bytes sent
    num += n
    cnt -= n
  }
  wait for 'SEND OK'
  return number of bytes sent if (num == len)
}
```

So if we hand simulate with len = 2050

initialize:
len = 2050
num = 0

loop 1:
cnt = 2048
n = 2048
num = 2048
cnt = 0
*** len is unchanged at 2050 ***

loop 2:
cnt = 2048 *** not 2 since len is never decremented ***
n = 2048 *** buffer overflow ***
num = 4096
cnt = 0
num != len so we loop forever
...

VladimirUmek, 2021-01-18:
You are absolutely right! Many thanks for insisting and providing additional explanation! For some reason I just didn't see it... 
I'll fix that.

xennex22, 2021-01-19:
My fix:

`uint32_t tx_size = len;`
...
`if (cnt > tx_size) {
     cnt = tx_size;
}`
...
`tx_size -= n;`

VladimirUmek, 2021-01-21:
The check within the loop was modified to
https://github.com/ARM-software/CMSIS-Driver/blob/7f340e089ef6b9adfce61046bf812a29684a3954/WiFi/ESP32/WiFi_ESP32.c#L3321-L3323
It should do the job.

xennex22, 2021-01-22:
I've confirmed the fix works for me.

closed 2021-01-22 by xennex22