ARM-software/CMSIS-Driver #8: WiFi GetNetInfo does not populate bssid

xennex22, 2021-01-22:
I'm fairly sure it should as the `ARM_WIFI_NET_INFO_t` structure contains a bssid member.  

It's fairly easy to add in after

https://github.com/ARM-software/CMSIS-Driver/blob/e549068aad6e377ecf0a67640b6e0812b9619e18/WiFi/ESP32/WiFi_ESP32.c#L2000

`memcpy(net_info->bssid, ap.bssid, sizeof net_info->bssid);`

VladimirUmek, 2021-01-26:
Hi,
member bssid does not exist in ARM_WIFI_NET_INFO_t, please see CMSIS-Driver WiFi API specification:
https://github.com/ARM-software/CMSIS_5/blob/5220bc2a76efeb290beb8aa523da5306983ba498/CMSIS/Driver/Include/Driver_WiFi.h#L119-L128

Member bssid does exist in ARM_WIFI_SCAN_INFO_s used to scan available WiFi networks.

xennex22, 2021-01-26:
Yes, you are correct.  I must have added it in because without it there is no way to match the AP you are connected to the scan list of APS if the SSID is hidden.

closed 2021-01-27 by xennex22