ARM-software/CMSIS-Driver #10: SPI_MultiSlave CS Support

Jonny-vb, 2021-04-15:
Hi - I'm trying to get some code working using the SPI_MultiSlave driver. However I am having some issues with controlling the CS lines. From the documentation it looks like all I should have to do is implement SPI_Control_SlaveSelect to set the correct IO for each device, however this only seems to get called once per call to Read/Write/Transfer functions and only ever seems to set the line high.

Is there something I'm missing here, or misunderstanding? I should say I'm currently only trying to control one device through the multislave driver (to be expanding when I get the one working!).

VladimirUmek, 2021-05-17:
Hi,
this looks like the problem with the documentation. I checked the driver implementation and its functionality relies on SPI.Control(ARM_SPI_CONTROL_SS, ...) handling. So code example in documentation is wrong and threads should be something like:
```c
/*----------------------------------------------------------------------------
 * SPI Thread A
 *---------------------------------------------------------------------------*/
__NO_RETURN static void Thread_A (void *argument) {
  char *p = "Sending data to Slave Device 0";
  (void)argument;
  
  SPI_A->Initialize(NULL);
  SPI_A->PowerControl(ARM_POWER_FULL);
  SPI_A->Control(ARM_SPI_MODE_MASTER | ARM_SPI_CPOL0_CPHA0  \
                                     | ARM_SPI_DATA_BITS(8) \
                                     | ARM_SPI_MSB_LSB      \
                                     | ARM_SPI_SS_MASTER_SW,
                                     10000000);
 
  SPI_A->Control(ARM_SPI_SET_DEFAULT_TX_VALUE, 0xFF);
 
  while(1) {
    /* Send to Slave Device 0 */
    SPI_A->Control (ARM_SPI_CONTROL_SS, ARM_SPI_SS_ACTIVE);
    SPI_A->Send (p, strlen(p));
    SPI_A->Control (ARM_SPI_CONTROL_SS, ARM_SPI_SS_INACTIVE);
 
    osDelay(10);
  }
}
 
/*----------------------------------------------------------------------------
 * SPI Thread B
 *---------------------------------------------------------------------------*/
__NO_RETURN static void Thread_B (void *argument) {
  char *p = "Sending data to Slave Device 1";
  (void)argument;
 
  /* Initialize and configure SPI */
  SPI_B->Initialize(NULL);
  SPI_B->PowerControl(ARM_POWER_FULL);
  SPI_B->Control(ARM_SPI_MODE_MASTER | ARM_SPI_CPOL1_CPHA1  \
                                     | ARM_SPI_DATA_BITS(8) \
                                     | ARM_SPI_MSB_LSB      \
                                     | ARM_SPI_SS_MASTER_SW,
                                     15000000);
 
  SPI_B->Control(ARM_SPI_SET_DEFAULT_TX_VALUE, 0xFF);
 
  while(1) {
    /* Send to Slave Device 1 */
    SPI_B->Control (ARM_SPI_CONTROL_SS, ARM_SPI_SS_ACTIVE);
    SPI_B->Send (p, strlen(p));
    SPI_B->Control (ARM_SPI_CONTROL_SS, ARM_SPI_SS_INACTIVE);
 
    osDelay(10);
  }
}
```
Each SPI instance is initialized with ARM_SPI_SS_MASTER_SW meaning that the driver Slave Select line is software controlled and before any transfer is initiated Control function shall be called with arguments ARM_SPI_CONTROL_SS and ARM_SPI_SS_ACTIVE (and paired with ARM_SPI_SS_INACTIVE when the transfer is completed). 
That should do the trick.

It is true that the underlying driver (the driver that is actually using the SPI controller peripheral) in configured as `ARM_SPI_SS_MASTER_UNUSED` but the statements in documentation are unclear and misleading especially when combined with the code example.
I'll make required corrections in the documentation.

Jonny-vb, 2021-05-17:
Hi @VladimirUmek, thanks for having a look at this.

I'm not quite sure that all adds up still...

Firstly, that would mean that the SPI_Control_SlaveSelect callback function is actually not needed? Or does setting ARM_SPI_SS_MASTER_SW at the multislave driver level use that function correctly? As the documentation states, using ARM_SPI_SS_MASTER_SW normally can only control a single SS line, which makes it not very useful for multislave!

Secondly, the documentation talks about this driver implementing a FIFO for requests, so that if the driver is currently busy processing another SPI, your request is queued and executed when possible:

> In such a case, transfer operations will be queued and executed immediately after the current transfer completes. The transfer queue operates as a FIFO, so transfers will be executed in the same call order as expected by the application.

With your changes to the example, this wouldn't work as the slave select lines wouldn't be synchronised with this delayed operations (in fact your updated example won't work anyway as the Send() function returns immediately, and does not wait to the transfer to complete, meaning you would be deactivating the SS signal almost as soon as it is activated).

VladimirUmek, 2021-05-17:
SPI_Control_SlaveSelect is still required. In case of a multislave driver in mode ARM_SPI_SS_MASTER_SW, SPI_Control_SlaveSelect is called instead of controlling SPI peripheral driven SS pin when using Control (ARM_SPI_CONTROL_SS, ...).

And, yes, you are right about missing wait for transfer complete event and FIFO requests. The transfers from other thread(s) are blocked until Control is called with ARM_SPI_CONTROL_SS, ARM_SPI_SS_INACTIVE. I'll correct this in the docs.

Assuming that the thread A is locking the driver and is transferring the data. If now the thread B calls the transfer function, that transfer is queued and second instance of the driver becomes busy. Thread B is now waiting until transfer complete event arrives and will keep waiting until the thread A calls Control (ARM_SPI_CONTROL_SS, ARM_SPI_SS_INACTIVE).

VladimirUmek, 2021-06-02:
Are corrections added in commit 88f07ca adequate or is there anything we should add?
I would like to keep things simple as possible so it might look brief, but it should contain all required info.

Jonny-vb, 2021-06-07:
Hi @VladimirUmek - I've not had a chance to test it thoroughly yet, but as per your updated documentation, using ARM_SPI_SS_MASTER_SW does causes things to work more as expected.

closed 2023-12-07 by VladimirUmek