ARM-software/CMSIS-Driver #30: STM32H743 USART can not set ARM_USART_EVENT_RX_TIMEOUT

1132520084, 2022-10-24:
STM32F429 is support ARM_USART_EVENT_RX_TIMEOUT , but I can't use it on STM32H743, Is this function under development or cannot be completed due to hardware limitations，We look forward to your reply

MiloradCvjetkovic, 2024-11-20:
Hi @1132520084, 

sorry for overlooking this issue earlier.

The USART driver for STMicroelectronics STM32 devices based on STM HAL is being developed here: https://github.com/Open-CMSIS-Pack/CMSIS-Driver_STM32/blob/main/Drivers/USART_STM32.c .

This driver does also not support `ARM_USART_EVENT_RX_TIMEOUT`.
Reason for that is that STM HAL functionality terminates the ongoing transfer if an timeout is detected, whereas CMSIS driver specification requires transfer to remain active and continue with just signaling the `ARM_USART_EVENT_RX_TIMEOUT` event.
Due to this functional incompatibility this feature is not planned to be implemented.

KeilChris, 2025-04-01:
Closing as "won't fix".

closed 2025-04-01 by KeilChris