ARM-software/CMSIS-Driver #38: Using CMSIS-Driver with ESP8266 on STM32H743XIH6 encounters errors

QingshuangZhu, 2025-01-20:
I'm experiencing an issue where `ARM_WIFI_PowerControl(ARM_POWER_FULL)` -> `SetupCommunication()` -> `WiFi_Wait()` returns -1, indicating a response timeout error, when using version 1.8.0. What could be the cause of this?

MiloradCvjetkovic, 2025-01-20:
Hi @QingshuangZhu, 

can you provide a little bit more information about your setup?

Do you have sparkfun ESP8266 WiFi module or something else?
Are you using an STM32H7 evaluation board or a custom hardware?

For some general suggestions you can check the the following [information](https://github.com/Open-CMSIS-Pack/NXP_IMXRT1050-EVKB_BSP/tree/main/Documents#wifi-shields), suggesting importance of proper RESET connection and external power supply of the shield.

QingshuangZhu, 2025-01-21:
Thank you very much for your reply.
I am using custom hardware along with the espressif ESP8266 WiFi module. As a beginner, I am not quite familiar with how to configure pinout using STM32 CubeMX. As shown in the figure below, the Power and Thermal settings have not been configured.

[image]

MiloradCvjetkovic, 2025-01-21:
Hi @QingshuangZhu,

thanks for the explanation, it is much clearer now.

Anyways, I would suggest you take a look at support for a similar board that we already have.

We have support for a board NUCLEO-H743ZI2 that has a H743 MCU (different variant than XIH6 but similar enough) and you can find that BSP development at the following GitHub repository https://github.com/Open-CMSIS-Pack/ST_NUCLEO-H743ZI2_BSP .

You can also find a BSP (Board Support Pack) pack at the following link: https://www.keil.arm.com/packs/nucleo-h743zi2_bsp-keil/overview/ .

What you will be interested in, is how this chip is configured by the STM32CubeMX so you can make similar configuration in your STM32CubeMX and the .ioc file that you can inspect is located in the previously mentioned GitHub repository, specifically at the following location https://github.com/Open-CMSIS-Pack/ST_NUCLEO-H743ZI2_BSP/blob/main/Layers/Default/CubeMX/STM32CubeMX/STM32CubeMX.ioc .

Now, this is all something that can help you, there is, however, a lot of information you have to digest, you can start with what are [CMSIS components](https://community.arm.com/arm-community-blogs/b/tools-software-ides-blog/posts/what-are-cmsis-software-components) and about [CMSIS-Toolbox and CubeMX generator](https://open-cmsis-pack.github.io/cmsis-toolbox/CubeMX/) .

QingshuangZhu, 2025-01-21:
Hi @MiloradCvjetkovic,
thank you very much for your help, i will follow your advice and learn.

closed 2025-01-22 by QingshuangZhu